itemisCREATE/statecharts #2163: Two different mouse cursor are confusing at transitions

rherrmannr, 2018-07-27:
The mouse cursor changes, if the mouse is exactly over the middle of a transition.

tkutz, 2018-07-30:
The different cursor indicates that you can select the transition now and specify a trigger. The cursor also changes when you hover a bend-point because in that position you can drag the bend-point to change transition routing.

BeckmaR, 2018-07-31:
I would agree, that's intentional. If there isn't anything completely going wrong on Windows, this one can be closed...

rherrmannr, 2018-08-01:
A different mouse cursor only appears in a very small area.

The mouse cursor changes in the red area, but got the same functionallity. 

[image]

BeckmaR, 2018-08-01:
I can confirm that this issue does **not** exist on Linux

tkutz, 2018-08-02:
Also *not* reproducible on macOS.

andreasmuelder, 2018-09-18:
external bug in GMF, we can't fix this here

closed 2018-09-18 by andreasmuelder