itemisCREATE/statecharts #2192: "Readonly" is deprecated but helpful

DefCon-CC, 2018-08-24:
In the latest Version's of Yakindu SCT the variable "prefix" readonly is marked as deprecated. But this is helpful if you would like to prevent that a variable can only be read from outside of the state machine and never been written.

terfloth, 2018-09-03:
Agree.

Check current implementation and remove deprecation.

rherrmannr, 2018-09-03:
@robert-rudi and I also discussed this. 

Readonly should not be marked as deprecated as introduced in https://github.com/Yakindu/statecharts/commit/7a11af1a4065a4f5bc1c3752a62d61a8df4fd7cd

rherrmannr, 2018-09-03:
The readonly keyword will be checked within every generator (C, Cpp, Java) for generating the sett functions: ``«IF !variable.readonly && !variable.const» ``.
So it's supported and not deprecated.

rherrmannr, 2018-09-04:
We will additionnally remove the deprecated keyword 'external'