itemisCREATE/statecharts #2203: check error/warning validation messages for expressiveness and quickfixes

tangele, 2018-08-30:
* in a usability test, we noticed that an user (who received the "missing trigger" warning) didn't knew where he was expected to add the missing trigger.
* in the case of the error message ”local declarations are not allowed in interface scope”, the same user was wondered how to fix that. He had defined an event without using the `in` or `out` keyword.

andreasmuelder, 2018-09-18:
We should check if we can replace the default xtext messages like
- missing EOF at 
- couldn't resolve reference to EObject
with more expressive messages

tangele, 2018-10-24:
additionally:

```
interface:
var test:boolean = valueof(noSuchName)
```

creates the error:
```
is no event.	default2.sct	/issue_2238	line: 21 /issue_2238/default2.sct	Statechart Check (Fast)
```