itemisCREATE/statecharts #2236: Statechart is dirty and can't be saved

rherrmannr, 2018-09-11:
Not sure how to reproduce this.

Statechart is dirty, save does not work or seems not to have an effect. Restart of IDE has no effect.

andreasmuelder, 2018-09-12:
Please reopen with steps to reproduce

svenjawendler, 2018-09-13:
[image]

How to reproduce:
Create a new C Project
Create a new C Statechart
Clear the DefinitionSection
- Paste the following: internal:
 event press:  
  var x: bool=false 
Add a second Statechart and transition to it with press / x=true

Errors occur, save the model, clean project, restart IDE -> no effect, problems stay

BeckmaR, 2018-09-14:
@svenjawendler this did not reproduce the issue for me. Also, your definition section just contains errors, the line `event press:` ends with a superfluos colon.

closed 2018-09-14 by BeckmaR