itemisCREATE/statecharts #2354: Testcase 4.16.1 and 4.16.2 does  not always work as intendet

rherrmannr, 2018-10-17:
Tried this together with @tkutz 

Sometimes the Statechart needs to be edited and saved.

tkutz, 2019-02-20:
4.18.1 does only not work initially after importing the project. This is because the new `First.h` resource is not yet known to the shared resource set first time after generating it. Once it was loaded, removing and re-generating it will properly unload all referencing resources.

tkutz, 2019-02-20:
4.18.2 also does not work the first time only; once the SCT is touched, and hence `First.h` is loaded, deleting it and re-generating will properly unload all referencing resources.

tkutz, 2019-02-20:
I cannot think of any simple solution to these issues. We would need to add quite some complexity just to fix a case that appears first time after importing a project.

Won't fix due to cost/benefit ratio.

closed 2019-02-20 by tkutz