itemisCREATE/statecharts #2370: Strg+C in the definition section causes InjectException

tangele, 2018-10-19:
try to copy/`Strg+C` something (what exactly doesn't matter).

```
!ENTRY org.eclipse.ui 4 0 2018-10-19 14:05:56.974
!MESSAGE Unhandled event loop exception
!STACK 0
org.eclipse.e4.core.di.InjectionException: java.lang.NullPointerException
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:65)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:161)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:152)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:494)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:204)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.executeCommand(KeyBindingDispatcher.java:305)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.press(KeyBindingDispatcher.java:579)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.processKeyEvent(KeyBindingDispatcher.java:648)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.filterKeySequenceBindings(KeyBindingDispatcher.java:438)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.access$2(KeyBindingDispatcher.java:381)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher$KeyDownFilter.handleEvent(KeyBindingDispatcher.java:93)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.filterEvent(Display.java:1190)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1051)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1076)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1061)
	at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1103)
	at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1099)
	at org.eclipse.swt.widgets.Widget.wmChar(Widget.java:1486)
	at org.eclipse.swt.widgets.Control.WM_CHAR(Control.java:4884)
	at org.eclipse.swt.widgets.Canvas.WM_CHAR(Canvas.java:350)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4765)
	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:345)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:4757)
	at org.eclipse.swt.internal.win32.OS.DispatchMessageW(Native Method)
	at org.eclipse.swt.internal.win32.OS.DispatchMessage(OS.java:2303)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3532)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1170)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1059)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:667)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:597)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:656)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:592)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1498)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1471)
Caused by: java.lang.NullPointerException
	at org.eclipse.papyrus.infra.gmfdiag.menu.handlers.AbstractGraphicalCommandHandler.execute(AbstractGraphicalCommandHandler.java:85)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:291)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:93)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	... 51 more

```

tangele, 2018-10-19:
TODO: test in release version

rherrmannr, 2018-10-19:
Please specify the used definition section. We got two:
- pinned
- unpinned

Is this an error for both?

rherrmannr, 2018-10-19:
Additionally, some keybindings do not work in the eclipse runtime.

Please verify if this occurs in the product.

tangele, 2018-10-19:
@rherrmannr what keybindings exactly?

rherrmannr, 2018-10-19:
I'm not sure if these are the same with windwos.
Just have a look at your console output, while starting a runtime.
For example.
```
!MESSAGE A conflict occurred for CTRL+SHIFT+X:
Binding(CTRL+SHIFT+X,
	ParameterizedCommand(Command(org.eclipse.papyrus.cut.diagram,Graphical Cut,
		Cut the elements in the current diagram,
		Category(org.eclipse.papyrus.editor.category,Papyrus Category,this a category of papyrus commands,true),
		org.eclipse.ui.internal.WorkbenchHandlerServiceHandler@61c9e646,
		,,true),null),
	org.eclipse.ui.defaultAcceleratorConfiguration,
	org.eclipse.gmf.runtime.diagram.ui.diagramContext,,,system)
```

BeckmaR, 2018-10-23:
Can't reproduce

tangele, 2018-10-23:
closing: papyrus defines some keybindings, but we only need the framework to draw the states from papyrus. As we also have keybindings, ours and papyrus' keybindings conflict, causing the observed exception. 

As we don't deploy SCT with the plugin containing the papyrus keybindings, this is not a bug.

closed 2018-10-23 by tangele