itemisCREATE/statecharts #2388: Only generate includes when used

andreasmuelder, 2018-10-26:
for C and C++, the headers 'stdlib.h' and 'string.h' are generated as includes into the C Module:
```
#include <stdlib.h>
#include <string.h>
```

We should only generate those includes if they are required

andreasmuelder, 2018-11-06:
Is it sufficient to always include the headers when the String / Char type is used in the statechart?

terfloth, 2018-11-20:
* <string.h> is just used if string compare is used in expressions. 
* <stdlib.h> seems not to be used. The initial assumption was that malloc & co is required for string concatenation. But this is not supported (type system constraint)

So the issue should be fixed in the following way for the C and C++ code generators independently:

* include <string.h> only if a string compare is required.
* check if there is somewhere (also C domain) a need for stdlib. If not remove it completely

terfloth, 2018-11-20:
@BeckmaR - please fix this issue.

PR #2440 (2018-11-21): Remove stdlib.h and string.h from C & C++
Close #2388
closed 2018-11-21 by terfloth