itemisCREATE/statecharts #2391: C-Code Generator should use proper suffixes for values

BeckmaR, 2018-10-29:
I just found a case where there's actually a difference between `1.7` and `1.7f`, such that the comparison `1.7 == 1.7f` actually returns false.

terfloth, 2018-11-20:
@BeckmaR : How is the comparison constructed? Is the default domain or the C domain used?

BeckmaR, 2018-11-26:
@terfloth SCTUnit uses the proper suffixes, and the open source C generator does not. I already implemented it in the C-Domain. In the default domain it's more complicated because we don't know what type a "real" is. I guess the absolutely correct behaviour would be to generate no suffix in all cases, except the user specifies one. But I think the literal parser doesn't keep those.

terfloth, 2018-12-10:
Our grammar defines DoubleLiteral and FloatLiteral. A literal that is not qualified with d/f will be one of both (i assume DoubleLiteral - must be checked). So we can't distinguish not qualified from double but from float (or vice versa). To have full control we would need an additional RealLiteral without f/d suffix.

- [ ] check if literals without suffix evaluate to FloatLiteral or DoubleLiteral
- [ ] check impact of additional RealLiteral to type inferrer and type systems implementation.

BeckmaR, 2019-02-14:
The default case is indeed a `DoubleLiteral`:
```
var noS : real = 1.7 => DoubleLiteral
var flS: real = 1.7f => FloatLiteral
var dS : real = 1.7d => DoubleLiteral
```

BeckmaR, 2019-02-14:
This is consistent to the C standard however, a number without suffix is a double literal. Java does the same, luckily.

PR #2689 (2019-02-27): Use f suffix for FloatLiterals
Resolve #2391
closed 2019-02-27 by BeckmaR