itemisCREATE/statecharts #2410: C Generator exception after changing domain (default -> SCXML) & removing definition section annotations

tangele, 2018-11-09:
1. create a new project
2. create a new statechart using the default domain.
3. create a YAKINDU SCT C Code Generator model using the above statechart.
4. change the statechart domain to SCXML
5. remove all annotations inside the definition section of the statechart
6. save the changes.
7. Exception occours.

version https://github.com/Yakindu/statecharts/commit/fdfa7f897f5a42f4be30b5bc1066d56cdecf7db3

```
'Execute SCT Genmodel default.sgen' has encountered a problem.

An error occurred during code generation.
Guice provision errors:

1) No implementation for org.yakindu.base.types.inferrer.ITypeSystemInferrer was bound.
  while locating org.yakindu.base.types.inferrer.ITypeSystemInferrer
    for field at org.yakindu.sct.generator.c.CExpressionsGenerator._iTypeSystemInferrer(Unknown Source)
  while locating org.yakindu.sct.generator.c.CExpressionsGenerator
    for field at org.yakindu.sct.generator.c.FlowCode._cExpressionsGenerator(Unknown Source)
  while locating org.yakindu.sct.generator.c.FlowCode
    for field at org.yakindu.sct.generator.c.submodules.APIGenerator._flowCode(Unknown Source)
  while locating org.yakindu.sct.generator.c.submodules.APIGenerator
    for field at org.yakindu.sct.generator.c.submodules.StatemachineHeaderFragment._aPIGenerator(Unknown Source)
  while locating org.yakindu.sct.generator.c.submodules.StatemachineHeaderFragment
  while locating org.yakindu.sct.generator.c.HeaderContentFragmentProvider
  while locating org.yakindu.sct.generator.c.files.StatemachineHeader
    for field at org.yakindu.sct.generator.c.CGenerator.statemachineHeader(Unknown Source)
  while locating org.yakindu.sct.generator.c.CGenerator
  while locating org.yakindu.sct.generator.core.IExecutionFlowGenerator
    for field at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.generator(Unknown Source)
  while locating org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor

2) No implementation for org.yakindu.base.types.inferrer.ITypeSystemInferrer was bound.
  while locating org.yakindu.base.types.inferrer.ITypeSystemInferrer
    for field at org.yakindu.sct.generator.c.CExpressionsGenerator._iTypeSystemInferrer(Unknown Source)
  while locating org.yakindu.sct.generator.c.CExpressionsGenerator
    for field at org.yakindu.sct.generator.c.FlowCode._cExpressionsGenerator(Unknown Source)
  while locating org.yakindu.sct.generator.c.FlowCode
    for field at org.yakindu.sct.generator.c.submodules.StatemachineSourceFragment._flowCode(Unknown Source)
  while locating org.yakindu.sct.generator.c.submodules.StatemachineSourceFragment
  while locating org.yakindu.sct.generator.c.SourceContentFragmentProvider
  while locating org.yakindu.sct.generator.c.files.StatemachineSource
    for field at org.yakindu.sct.generator.c.CGenerator.statemachineSource(Unknown Source)
  while locating org.yakindu.sct.generator.c.CGenerator
  while locating org.yakindu.sct.generator.core.IExecutionFlowGenerator
    for field at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.generator(Unknown Source)
  while locating org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor

3) No implementation for org.yakindu.base.types.inferrer.ITypeSystemInferrer was bound.
  while locating org.yakindu.base.types.inferrer.ITypeSystemInferrer
    for field at org.yakindu.sct.generator.c.CExpressionsGenerator._iTypeSystemInferrer(Unknown Source)
  while locating org.yakindu.sct.generator.c.ConstantInitializationResolver
    for field at org.yakindu.sct.generator.c.submodules.StatemachineSourceFragment._constantInitializationResolver(Unknown Source)
  while locating org.yakindu.sct.generator.c.submodules.StatemachineSourceFragment
  while locating org.yakindu.sct.generator.c.SourceContentFragmentProvider
  while locating org.yakindu.sct.generator.c.files.StatemachineSource
    for field at org.yakindu.sct.generator.c.CGenerator.statemachineSource(Unknown Source)
  while locating org.yakindu.sct.generator.c.CGenerator
  while locating org.yakindu.sct.generator.core.IExecutionFlowGenerator
    for field at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.generator(Unknown Source)
  while locating org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor

4) No implementation for org.yakindu.base.types.inferrer.ITypeSystemInferrer was bound.
  while locating org.yakindu.base.types.inferrer.ITypeSystemInferrer
    for field at org.yakindu.sct.generator.c.CExpressionsGenerator._iTypeSystemInferrer(Unknown Source)
  while locating org.yakindu.sct.generator.c.CExpressionsGenerator
    for field at org.yakindu.sct.generator.c.extensions.ExpressionsChecker._cExpressionsGenerator(Unknown Source)
  while locating org.yakindu.sct.generator.c.extensions.ExpressionsChecker
[... 58 lines cut ...]
  while locating org.yakindu.sct.generator.core.IExecutionFlowGenerator
    for field at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.generator(Unknown Source)
  while locating org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor

8) No implementation for org.yakindu.base.types.inferrer.ITypeSystemInferrer was bound.
  while locating org.yakindu.base.types.inferrer.ITypeSystemInferrer
    for field at org.yakindu.sct.generator.c.CExpressionsGenerator._iTypeSystemInferrer(Unknown Source)
  while locating org.yakindu.sct.generator.c.CExpressionsGenerator
    for field at org.yakindu.sct.generator.c.extensions.ExpressionsChecker._cExpressionsGenerator(Unknown Source)
  while locating org.yakindu.sct.generator.c.extensions.ExpressionsChecker
    for field at org.yakindu.sct.generator.c.submodules.InternalFunctionsGenerator._expressionsChecker(Unknown Source)
  while locating org.yakindu.sct.generator.c.submodules.InternalFunctionsGenerator
    for field at org.yakindu.sct.generator.c.submodules.StatemachineSourceFragment._internalFunctionsGenerator(Unknown Source)
  while locating org.yakindu.sct.generator.c.submodules.StatemachineSourceFragment
  while locating org.yakindu.sct.generator.c.SourceContentFragmentProvider
  while locating org.yakindu.sct.generator.c.files.StatemachineSource
    for field at org.yakindu.sct.generator.c.CGenerator.statemachineSource(Unknown Source)
  while locating org.yakindu.sct.generator.c.CGenerator
  while locating org.yakindu.sct.generator.core.IExecutionFlowGenerator
    for field at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.generator(Unknown Source)
  while locating org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor

8 errors

```

tangele, 2018-11-09:
ref https://github.com/Yakindu/sctpro/issues/675

not a duplicate of the above; the generator should not even start of an invalid statechart.

BeckmaR, 2019-02-11:
The C Generator does not support the SCXML domain at all. This should of course be communicated to the user, but this is a pro issue as the scxml domain is pro.

BeckmaR, 2019-07-15:
Wrong location, closing

closed 2019-07-15 by BeckmaR