itemisCREATE/statecharts #2444: Add support for external event queues for event driven C++ code generator

terfloth, 2018-11-21:
The C++ code generator currently does not support external event queues. Add this feature to the code generator according to the following requirements:

- [ ] The statechart class will contain an additional queue. Use std::deque for the implementation.
- [ ] The implementation should follow the same implementation pattern like the internal event queue.
- [ ] The raise methods queue the events instead of directly invoking the runCycle.
- [ ] A run loop method is required that processes the events on the event queue.

Open points:
- [ ]  Define if event raising automaticall should trigger the run loop or if that should be called separately.

BeckmaR, 2018-11-21:
How do we handle out events? Say I queue `e1`, `e2`, `e3`, and run the loop. During the runs, the out events `o1`, `o2` and `o3` are raised, respectively. Will they be queued as well? Just leaving them as active might lead to wrong behavior if they are used in a condition.

How are internal events merged with this? Do we loop through in-events in the outer loop, and through internal events in an inner loop?

terfloth, 2018-11-27:
First step is implementing a derived C++ code generator that add the event queue for in events to the state machine. In a second step this feature must be merged with the original C++ code generator.

BeckmaR, 2019-01-16:
Closed via #2577

closed 2019-01-16 by BeckmaR