itemisCREATE/statecharts #2500: Improve moving transitions

rherrmannr, 2018-12-05:
The behaviour while moving transitions is annoying. Transitions are always settled in the middle of the state. 

This is shown in the picture, where I have moved a transition from the right side of the state to the upper side. In my opinion, the transition should be placed in the right corner of the upper side.

[image]

BeckmaR, 2018-12-11:
In my opinion, there should be "anchors" to which transitions can connect (and nowhere else). There should be one in the middle and one in every corner, and with increasing state size there should be more anchors.

andreasmuelder, 2020-03-16:
handled with new router/layouter

closed 2020-03-16 by andreasmuelder