itemisCREATE/statecharts #2520: Syntax highlighting in YAKINDU SCT not correct

WolfgangHa, 2018-12-12:
In a transition condition the syntax highlighting is wrong if an expression appears commented and uncommented.

Transition not in edit mode:
[image]
Transition in edit mode:
[image]

OBSERVED: The expression "a >= 1" appears in color for commented code.
EXPECTED: The expression "a >= 1" should NOT appear in color for commented code.

WolfgangHa, 2018-12-12:
We assume that the bug was introduced with: 
https://github.com/Yakindu/statecharts/pull/1709

andreasmuelder, 2018-12-13:
It seems like the lineOffset in SyntaxColoringLabel#paintText is calculated wrong when the same text appears multiple times in different lines sind indexOf returns the first occurence:

`int lineOffset = getText().indexOf(originalDraw);`

commit d3478471 (2019-02-25): fixes #2520
commit b425b81d (2019-02-25): fixes #2520 (#2682)
closed 2019-02-25 by andreasmuelder