itemisCREATE/statecharts #2560: installation does not complete - neon - Linux

artbody, 2019-01-05:
Hy there
on a Gentoo Linux

up to here its installing fine.
15 (Optional) While installing, clicking on the double-arrow (double arrow) allows you to see the further progression. 
Execution startup task runs to 73%
```
Importing project org.yakindu.base.types.edit
Importing project org.yakindu.sct.generator.genmodel
Performing Target Platform Neon (SCT)
Fetching content.jar from http://download.itemis.com/updates/releases/2.1.1/ (5,47kB)
Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/
Retrieving data from http://marketplace.eclipse.org
Fetching compositeContent.jar from http://download.eclipse.org/eclipse/updates/4.6/ (476B)
Retrieving data from http://marketplace.eclipse.org
Fetching content.jar from http://download.eclipse.org/eclipse/updates/4.6/categoriesNeon/ (1,46kB)
Retrieving data from http://marketplace.eclipse.org
Connecting Git team provider.
Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/R-4.6-201606061100/
Retrieving data from http://marketplace.eclipse.org
Fetching content.xml.xz from http://download.eclipse.org/eclipse/updates/4.6/R-4.6-201606061100/
Retrieving data from http://marketplace.eclipse.org
Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.1-201609071200/
Fetching content.xml.xz from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.1-201609071200/
Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.2-201611241400/
Fetching content.xml.xz from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.2-201611241400/
Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.3-201703010400/
Fetching content.xml.xz from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.3-201703010400/
Fetching p2.index from http://download.eclipse.org/releases/neon/
Fetching compositeContent.jar from http://download.eclipse.org/releases/neon/ (518B)
Fetching p2.index from http://download.eclipse.org/technology/epp/packages/neon/
Fetching content.xml.xz from http://download.eclipse.org/technology/epp/packages/neon/
Fetching p2.index from http://download.eclipse.org/releases/neon/201705151400/
Fetching content.xml.xz from http://download.eclipse.org/releases/neon/201705151400/
Fetching p2.index from http://download.eclipse.org/releases/neon/201703141400/
Fetching content.xml.xz from http://download.eclipse.org/releases/neon/201703141400/
Searching for markers
Process resource updates
Fetching p2.index from http://download.eclipse.org/releases/neon/201703231000/
Fetching p2.index from http://download.eclipse.org/tools/orbit/downloads/drops/R20140525021250/repository/
Fetching content.xml.xz from http://download.eclipse.org/tools/orbit/downloads/drops/R20140525021250/repository/
Fetching content.jar from http://download.eclipse.org/egit/updates-4.5/ (27,02kB)
ERROR: org.eclipse.pde.core code=0 Problems occurred while resolving the target contents
  at org.eclipse.oomph.util.OomphPlugin.coreException(OomphPlugin.java:280)
  at org.eclipse.oomph.util.pde.TargetPlatformUtil.activateTargetDefinition(TargetPlatformUtil.java:151)
  at org.eclipse.oomph.setup.pde.impl.TargetPlatformTaskImpl.perform(TargetPlatformTaskImpl.java:293)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.doPerformNeededSetupTasks(SetupTaskPerformer.java:3334)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.access$1(SetupTaskPerformer.java:3277)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil$1.run(SetupTaskPerformer.java:4504)
  at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2295)
  at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2322)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.performNeededSetupTasks(SetupTaskPerformer.java:4498)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.access$0(SetupTaskPerformer.java:4496)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performNeededSetupTasks(SetupTaskPerformer.java:3268)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performTriggeredSetupTasks(SetupTaskPerformer.java:3243)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.perform(SetupTaskPerformer.java:3143)
  at org.eclipse.oomph.setup.ui.wizards.ProgressPage$9.run(ProgressPage.java:575)
  at org.eclipse.oomph.setup.ui.wizards.ProgressPage$11$1.run(ProgressPage.java:701)
  at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
  ERROR: org.eclipse.pde.core code=0 Problems loading repositories
    ERROR: org.eclipse.pde.core code=0 Problems loading repositories
      ERROR: org.eclipse.equinox.p2.metadata.repository code=1002 Unable to read repository at http://download.eclipse.org/releases/neon.
      ERROR: org.eclipse.equinox.p2.transport.ecf code=1002 Unable to read repository at http://download.eclipse.org/releases/neon/201703231000/content.xml.xz.
      java.net.SocketException: Connection reset
        at java.net.SocketInputStream.read(SocketInputStream.java:210)
        at java.net.SocketInputStream.read(SocketInputStream.java:141)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.fillBuffer(AbstractSessionInputBuffer.java:161)
        at org.apache.http.impl.io.SocketInputBuffer.fillBuffer(SocketInputBuffer.java:82)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.readLine(AbstractSessionInputBuffer.java:278)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:138)
[... 78 lines cut ...]
        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
        at java.net.Socket.connect(Socket.java:589)
        at org.eclipse.ecf.internal.provider.filetransfer.httpclient4.ECFHttpClientProtocolSocketFactory.connectSocket(ECFHttpClientProtocolSocketFactory.java:86)
        at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:180)
        at org.apache.http.impl.conn.AbstractPoolEntry.open(AbstractPoolEntry.java:144)
        at org.apache.http.impl.conn.AbstractPooledConnAdapter.open(AbstractPooledConnAdapter.java:134)
        at org.apache.http.impl.client.DefaultRequestDirector.tryConnect(DefaultRequestDirector.java:610)
        at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:445)
        at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:835)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83)
        at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer.performConnect(HttpClientRetrieveFileTransfer.java:1084)
        at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer.access$0(HttpClientRetrieveFileTransfer.java:1075)
        at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer$1.performFileTransfer(HttpClientRetrieveFileTransfer.java:1071)
        at org.eclipse.ecf.filetransfer.FileTransferJob.run(FileTransferJob.java:74)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

Took 619 seconds.
There are failed tasks.
Press Back to choose different settings or Cancel to abort.

```
something strange 
i think there are some files missing because it needs a lot of time (10min+),
 from here` neon/201703231000/ `
until there is a timeout

artbody, 2019-01-05:
i tried to install it manually 
```
Unable to read repository at http://download.eclipse.org/releases/neon/.
Unable to read repository at http://download.eclipse.org/releases/neon/.
Unable to read repository at http://download.eclipse.org/releases/neon/201703231000/content.xml.xz.
Connection reset
```
```

ava.net.SocketTimeoutException: connect timed out
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at org.eclipse.ecf.internal.provider.filetransfer.httpclient4.ECFHttpClientProtocolSocketFactory.connectSocket(ECFHttpClientProtocolSocketFactory.java:86)
	at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:180)
	at org.apache.http.impl.conn.AbstractPoolEntry.open(AbstractPoolEntry.java:144)
	at org.apache.http.impl.conn.AbstractPooledConnAdapter.open(AbstractPooledConnAdapter.java:134)
	at org.apache.http.impl.client.DefaultRequestDirector.tryConnect(DefaultRequestDirector.java:610)
	at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:445)
	at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:835)
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer.performConnect(HttpClientRetrieveFileTransfer.java:1084)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer.access$0(HttpClientRetrieveFileTransfer.java:1075)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientRetrieveFileTransfer$1.performFileTransfer(HttpClientRetrieveFileTransfer.java:1071)
	at org.eclipse.ecf.filetransfer.FileTransferJob.run(FileTransferJob.java:74)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

```

artbody, 2019-01-05:
```
eclipse.buildId=4.10.0.I20181206-0815
java.version=1.8.0_192
java.vendor=Oracle Corporation
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=de_DE
Framework arguments:  -product org.eclipse.epp.package.committers.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.committers.product

```
`Unable to read repository at http://download.eclipse.org/releases/neon/201703231000/content.xml.xz.`

i think this is a bug

artbody, 2019-01-05:
```
.................
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1474)
!SUBENTRY 1 org.eclipse.ui.monitoring 1 0 2019-01-05 10:46:46.677
!MESSAGE Sample at 10:46:46.662 (+0,459s)
Thread 'main' tid=1 (RUNNABLE)
!STACK 0
Stack Trace
	at com.google.common.collect.Iterators.addAll(Iterators.java:367)
	at com.google.common.collect.Lists.newArrayList(Lists.java:165)
	at org.eclipse.xtext.EcoreUtil2.getAllContentsOfType(EcoreUtil2.java:188)
	at org.eclipse.xtext.ui.editor.templates.XtextTemplateContextTypeRegistry.registerContextTypes(XtextTemplateContextTypeRegistry.java:50)
	at org.eclipse.xtext.ui.editor.templates.XtextTemplateContextTypeRegistry.<init>(XtextTemplateContextTypeRegistry.java:39)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at com.google.inject.internal.DefaultConstructionProxyFactory$1.newInstance(DefaultConstructionProxyFactory.java:85)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:85)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:254)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1031)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
	at com.google.inject.Scopes$1$1.get(Scopes.java:65)
	at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:40)
	at com.google.inject.internal.FactoryProxy.get(FactoryProxy.java:54)
	at com.google.inject.internal.SingleFieldInjector.inject(SingleFieldInjector.java:53)
	at com.google.inject.internal.MembersInjectorImpl.injectMembers(MembersInjectorImpl.java:110)
	at com.google.inject.internal.ConstructorInjector.construct(ConstructorInjector.java:94)
	at com.google.inject.internal.ConstructorBindingImpl$Factory.get(ConstructorBindingImpl.java:254)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter$1.call(ProviderToInternalFactoryAdapter.java:46)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1031)
	at com.google.inject.internal.ProviderToInternalFactoryAdapter.get(ProviderToInternalFactoryAdapter.java:40)
	at com.google.inject.Scopes$1$1.get(Scopes.java:65)
	at com.google.inject.internal.InternalFactoryToProviderAdapter.get(InternalFactoryToProviderAdapter.java:40)
	at com.google.inject.internal.InternalInjectorCreator$1.call(InternalInjectorCreator.java:204)
	at com.google.inject.internal.InternalInjectorCreator$1.call(InternalInjectorCreator.java:198)
	at com.google.inject.internal.InjectorImpl.callInContext(InjectorImpl.java:1024)
	at com.google.inject.internal.InternalInjectorCreator.loadEagerSingletons(InternalInjectorCreator.java:198)
	at com.google.inject.internal.InternalInjectorCreator.injectDynamically(InternalInjectorCreator.java:179)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:109)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at org.eclipse.xtend.ide.internal.XtendActivator.createInjector(XtendActivator.java:71)
	at org.eclipse.xtend.ide.internal.XtendActivator.getInjector(XtendActivator.java:59)
	at org.eclipse.xtend.ide.XtendExecutableExtensionFactory.getInjector(XtendExecutableExtensionFactory.java:30)
	at org.eclipse.xtext.ui.guice.AbstractGuiceAwareExecutableExtensionFactory.create(AbstractGuiceAwareExecutableExtensionFactory.java:53)
	at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:265)
	at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:63)
	at org.eclipse.ui.internal.console.PatternMatchListenerExtension.createDelegate(PatternMatchListenerExtension.java:84)
	at org.eclipse.ui.internal.console.PatternMatchListener.<init>(PatternMatchListener.java:29)
	at org.eclipse.ui.internal.console.ConsoleManager.createPatternMatchListeners(ConsoleManager.java:414)
	at org.eclipse.ui.internal.console.ConsoleManager.addConsoles(ConsoleManager.java:219)
	at org.eclipse.jdt.debug.ui.console.JavaStackTraceConsoleFactory.openConsole(JavaStackTraceConsoleFactory.java:72)
	at org.eclipse.ui.internal.views.log.jdt.ShowErrorInStackTraceConsoleHandler.execute(ShowErrorInStackTraceConsoleHandler.java:61)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:294)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:96)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[... cut ...]
	at org.apache.felix.scr.impl.ComponentActorThread.run(ComponentActorThread.java:81)
	at java.lang.Thread.run(Thread.java:748)
!SUBENTRY 2 org.eclipse.ui.monitoring 1 0 2019-01-05 10:46:46.678
!MESSAGE Thread 'Worker-22: File Transport Cancel Handler' tid=62 (TIMED_WAITING)
!STACK 0
Stack Trace
	at java.lang.Thread.sleep(Native Method)
	at org.eclipse.equinox.internal.p2.transport.ecf.FileReader$CancelHandler.run(FileReader.java:157)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
```

tangele, 2019-01-06:
@artbody Thank you, we will look into it.

BeckmaR, 2019-01-07:
Hey, you might want to just give it a second chance. From time to time, the eclipse repositories / mirrors happen to be unreachable, and trying again after 12 - 24h can be enough to resolve the issue.

andreasmuelder, 2019-01-07:
@artbody as @BeckmaR already said, it looks like the eclipse mirror was down at that time. Can you retry?

tangele, 2019-01-08:
> Unable to read repository at http://download.eclipse.org/releases/neon/201703231000/content.xml.xz.

Seems fishy. (notice the dot at the end)

`wget http://download.eclipse.org/releases/neon/201703231000/content.xml.xz.` returns 404, but
`wget http://download.eclipse.org/releases/neon/201703231000/content.xml.xz` is successful. Therefore, i thought this is **not** an external bug.

BeckmaR, 2019-02-05:
Closing because of inactivity

closed 2019-02-05 by BeckmaR