itemisCREATE/statecharts #2614: install Error

XiaoXiaoYi123, 2019-01-17:
when I try to install the software as the readme tells,
there are some errors like this ,
i don't know how to resolve,
so i'd like some one to help me.
Thank you very much.

Performing Launch GenerateExpressions.mwe2
java.lang.Exception: No launcher found for GenerateExpressions.mwe2
  at org.eclipse.oomph.setup.launching.impl.LaunchTaskImpl.perform(LaunchTaskImpl.java:262)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.doPerformNeededSetupTasks(SetupTaskPerformer.java:3334)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.access$1(SetupTaskPerformer.java:3277)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil$1.run(SetupTaskPerformer.java:4504)
  at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2295)
  at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2322)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.performNeededSetupTasks(SetupTaskPerformer.java:4498)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.access$0(SetupTaskPerformer.java:4496)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performNeededSetupTasks(SetupTaskPerformer.java:3268)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performTriggeredSetupTasks(SetupTaskPerformer.java:3243)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.perform(SetupTaskPerformer.java:3143)
  at org.eclipse.oomph.setup.ui.wizards.ProgressPage$9.run(ProgressPage.java:575)
  at org.eclipse.oomph.setup.ui.wizards.ProgressPage$11$1.run(ProgressPage.java:701)
  at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

There are failed tasks.

BeckmaR, 2019-01-24:
Hey, I just tried and I failed to reproduce this. Please make sure you followed the given instructions exactly and try again. Also check if you have JDK version 8, 9 or 10, 8 would be ideal. If it still fails, please post more information about what you did and about your system.

andreasmuelder, 2019-01-30:
please reopen if this still exists

closed 2019-01-30 by andreasmuelder