itemisCREATE/statecharts #2638: Following build instructions in README does not give a functional install

Kilill, 2019-01-30:
Neither setup instructions for OHMP or without OHMP results in an error free functional install
OS: Linux Ubuntu 16.04
Java: Oracle 1.8.0_201

Not being a eclipse developer i have no clue how to create . copy appropriate build logs...

Kilill, 2019-01-30:
redid the hole Oohmp dance once more , it completed the setup, started a new eclipse instance, after the build clean step there are about 5-6000 errors... 
took a look at the errors and it seems that there are a lot of unresolved imports:
The import com.google.inject.multibindings cannot be resolved
The import org.eclipse.core.commands cannot be resolved
The import org.eclipse.core.resources cannot be resolved
The import org.eclipse.core.runtime cannot be resolved
The import org.eclipse.core.runtime.CoreException cannot be resolved
The import org.eclipse.core.runtime.Path cannot be resolved
The import org.eclipse.draw2d cannot be resolved
The import org.eclipse.emf.compare cannot be resolved
The import org.eclipse.emf.transaction cannot be resolved
The import org.eclipse.emf.workspace cannot be resolved
The import org.eclipse.gef cannot be resolved
The import org.eclipse.gef.commands cannot be resolved
The import org.eclipse.gef.ConnectionEditPart cannot be resolved
The import org.eclipse.gef.ContextMenuProvider cannot be resolved
The import org.eclipse.gef.DragTracker cannot be resolved
The import org.eclipse.gef.EditDomain cannot be resolved
The import org.eclipse.gef.EditPart cannot be resolved
The import org.eclipse.gef.editparts cannot be resolved
The import org.eclipse.gef.EditPartViewer cannot be resolved
The import org.eclipse.gef.editpolicies cannot be resolved
The import org.eclipse.gef.EditPolicy cannot be resolved
The import org.eclipse.gef.GraphicalEditPart cannot be resolved
The import org.eclipse.gef.GraphicalViewer cannot be resolved
The import org.eclipse.gef.Handle cannot be resolved
The import org.eclipse.gef.handles cannot be resolved
The import org.eclipse.gef.KeyHandler cannot be resolved
The import org.eclipse.gef.KeyStroke cannot be resolved
The import org.eclipse.gef.palette cannot be resolved
The import org.eclipse.gef.Request cannot be resolved
The import org.eclipse.gef.RequestConstants cannot be resolved
The import org.eclipse.gef.requests cannot be resolved
The import org.eclipse.gef.RootEditPart cannot be resolved
The import org.eclipse.gef.SnapToGeometry cannot be resolved
The import org.eclipse.gef.SnapToGuides cannot be resolved
The import org.eclipse.gef.Tool cannot be resolved
The import org.eclipse.gef.tools cannot be resolved
The import org.eclipse.gef.ui cannot be resolved
The import org.eclipse.gmf cannot be resolved
The import org.eclipse.jface.text cannot be resolved
The import org.eclipse.jface.viewers cannot be resolved
The import org.eclipse.swt cannot be resolved
The import org.eclipse.swt.graphics cannot be resolved
The import org.eclipse.swt.widgets cannot be resolved
The import org.eclipse.ui cannot be resolved
The import org.eclipse.ui.forms cannot be resolved
The import org.eclipse.ui.views.properties.tabbed cannot be resolved

BeckmaR, 2019-02-04:
Hi @Kilill, sorry I did not answer earlier.

This sounds a lot like the target couldn't be set / resolved correctly. In that case, you need to navigate to the plugin `org.yakindu.base.target` in the package explorer, open `2018-09.target` and in the upper right of the target editor choose _Set as Active Target Platform_. This will take some time for sure. Afterwards you'll need to generate the Xtext grammars:
* Rightclick on org.yakindu.base.expressions/Expressions.xtext => Run As.. => Generate Xtext Artifacts 
* Rightclick on org.yakindu.sct.stext/SText.xtext => Run As.. => Generate Xtext Artifacts
* Rightclick org.yakindu.sct.generator.genmodel/SGen.xtext => Run As.. => Generate Xtext Artifacts

BeckmaR, 2019-02-05:
This issue arose because of a faulty target platform file. If you follow the steps mentioned above, everything should work:, alternatively, you could try do the oomph-setup again. I consider this issue fixed via #2651 - if you encounter any issues again, feel free to open a new issue or re-open this issue again.

closed 2019-02-05 by BeckmaR