itemisCREATE/statecharts #2674: Cannot delete states or transitions

ankls, 2019-02-15:
States or transitions that are added to a chart cannot be removed again.

What did I try:
Selecting the state or transition and then
a) The usual keys (delete, ctrl-x): No reaction
b) The "Edit" menu: Cut, Delete etc. are greyed out (disabled). Available are Undo/Redo/Paste/FindReplace/Bookmark.

The bug could be related to #340.

Platform is Windows 7
Yakindu SCT v3.5.0
Eclipse 9.5.3.201909121146

tkutz, 2019-02-15:
Thanks for the bug report, we will have a look.

A few questions up-front:
* Do you have the same problems with newly created statecharts, or only with one specific statechart?
* Do the problems remain when you switch to a new workspace (File -> Switch Workspace..)?
* Are you using the product bundle from our download page, or did you install SCT into another IDE?

ankls, 2019-02-15:
Thanks for looking into it!

- The statecharts are created from scratch by me, and so far it did occur all the times (3 attempts, I'm new to this tool)
- The problem vanishes when I switch to the (currently only, hence same) workspace using File -> Switch Workspace.
- The installation was a complete bundle, not a pre-existing Eclipse

Your hint with "switch workspace" is a workaround, nice!

tkutz, 2019-02-15:
We still want to understand why these problems come up. 
Can you send us an error log, once you encounter the same problem again? You can open the error log by via "About YAKINDU SCT -> Installation Details -> Configuration -> View Error Log". The log might contain some hints for us. 

Thanks!

ankls, 2019-02-15:
I'm currently struggeling to get the erroneus situation back again... but nevertheless, here's the log:

[sct_error_log.txt](https://github.com/Yakindu/statecharts/files/2868679/sct_error_log.txt)

When I can get back to the situation, I'll add another one.

ankls, 2019-02-15:
The issue reoccurred while editing. I saved an error log, but that one is binary identical to the first one. 

Then I tried "switch to workspace" again, the tool restarted, and the issue was gone again.

tkutz, 2019-02-15:
Okay thanks for checking!

One interesting part in the error log is this exception here:
```
!ENTRY org.eclipse.core.databinding.observable 4 2 2019-02-14 17:36:31.814
!MESSAGE Problems occurred when invoking code from plug-in: "org.eclipse.core.databinding.observable".
!STACK 0
java.lang.NullPointerException
	at org.eclipse.gmf.runtime.common.ui.resources.FileModificationValidator$ModificationStamp.<init>(FileModificationValidator.java:279)
	at org.eclipse.gmf.runtime.common.ui.resources.FileModificationValidator.validateEdit(FileModificationValidator.java:213)
	at org.yakindu.sct.ui.editor.propertysheets.ValidatingEMFDatabindingContext.isWriteable(ValidatingEMFDatabindingContext.java:42)
	at org.yakindu.sct.ui.editor.propertysheets.ValidatingEMFDatabindingContext$1.doSet(ValidatingEMFDatabindingContext.java:52)
	at org.eclipse.core.databinding.ValueBinding.lambda$1(ValueBinding.java:200)
	at org.eclipse.core.databinding.observable.Realm$1.run(Realm.java:152)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.databinding.observable.Realm.safeRun(Realm.java:156)
	at org.eclipse.core.databinding.observable.Realm.exec(Realm.java:174)
	at org.eclipse.core.databinding.ValueBinding.lambda$0(ValueBinding.java:193)
	at org.eclipse.core.databinding.observable.Realm$1.run(Realm.java:152)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.databinding.observable.Realm.safeRun(Realm.java:156)
	at org.eclipse.core.databinding.observable.Realm.exec(Realm.java:174)
	at org.eclipse.core.databinding.ValueBinding.doUpdate(ValueBinding.java:162)
	at org.eclipse.core.databinding.ValueBinding.access$4(ValueBinding.java:151)
	at org.eclipse.core.databinding.ValueBinding$1.handleValueChange(ValueBinding.java:47)
	at org.eclipse.core.databinding.observable.value.ValueChangeEvent.dispatch(ValueChangeEvent.java:73)
	at org.eclipse.core.databinding.observable.ChangeManager.fireEvent(ChangeManager.java:120)
	at org.eclipse.core.databinding.observable.value.DecoratingObservableValue.fireValueChange(DecoratingObservableValue.java:64)
	at org.eclipse.core.databinding.observable.value.DecoratingObservableValue.handleValueChange(DecoratingObservableValue.java:101)
	at org.eclipse.core.databinding.observable.value.DecoratingObservableValue.lambda$0(DecoratingObservableValue.java:76)
	at org.eclipse.core.databinding.observable.value.ValueChangeEvent.dispatch(ValueChangeEvent.java:73)
	at org.eclipse.core.databinding.observable.ChangeManager.fireEvent(ChangeManager.java:120)
	at org.eclipse.core.databinding.observable.value.AbstractObservableValue.fireValueChange(AbstractObservableValue.java:85)
	at org.eclipse.core.internal.databinding.property.value.SimplePropertyObservableValue.notifyIfChanged(SimplePropertyObservableValue.java:119)
	at org.eclipse.core.internal.databinding.property.value.SimplePropertyObservableValue.lambda$1(SimplePropertyObservableValue.java:68)
	at org.eclipse.core.databinding.observable.Realm$1.run(Realm.java:152)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.databinding.observable.Realm.safeRun(Realm.java:156)
	at org.eclipse.core.databinding.observable.Realm.exec(Realm.java:174)
	at org.eclipse.core.internal.databinding.property.value.SimplePropertyObservableValue.lambda$0(SimplePropertyObservableValue.java:66)
	at org.eclipse.core.databinding.property.NativePropertyListener.fireChange(NativePropertyListener.java:72)
	at org.eclipse.jface.internal.databinding.swt.WidgetListener.handleEvent(WidgetListener.java:59)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4145)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1055)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1060)
	at org.eclipse.swt.widgets.Control.sendFocusEvent(Control.java:2958)
	at org.eclipse.swt.widgets.Widget.wmKillFocus(Widget.java:1817)
	at org.eclipse.swt.widgets.Control.WM_KILLFOCUS(Control.java:5168)
	at org.eclipse.swt.widgets.Canvas.WM_KILLFOCUS(Canvas.java:429)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4802)
	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:348)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:4792)
	at org.eclipse.swt.internal.win32.OS.SetFocus(Native Method)
	at org.eclipse.swt.widgets.Control.forceFocus(Control.java:1116)
	at org.eclipse.gef.ui.parts.DomainEventDispatcher.dispatchMousePressed(DomainEventDispatcher.java:346)
	at org.eclipse.draw2d.LightweightSystem$EventHandler.mouseDown(LightweightSystem.java:523)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:196)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4145)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1055)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3958)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3561)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
[... cut ...]
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
```
We will check if this is causing the effect that you are observing.

BeckmaR, 2019-02-18:
I could not find a connection between an NPE there and transitions that could not be deleted. It just makes it impossible to save the statechart after the definition section was edited.

BeckmaR, 2019-02-18:
I found another interesting section in the error log:
```
!ENTRY org.eclipse.emf.transaction 2 100 2019-02-07 16:31:30.596
!MESSAGE Resource was re-loaded while it was being unloaded, probably because of unintentional proxy resolution: platform:/resource/CoarseClock/Worker.sct

!ENTRY org.eclipse.emf.transaction 4 41 2019-02-07 16:31:30.627
!MESSAGE Read-only transaction rolled back
!SUBENTRY 1 org.eclipse.emf.transaction 4 40 2019-02-07 16:31:30.627
!MESSAGE Transaction aborted due to concurrent write

!ENTRY org.eclipse.ui 4 0 2019-02-07 16:31:31.361
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.NullPointerException
        at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.selectionChanged(CompatibilityPart.java:473)
        at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramGraphicalViewer.flushSelectionEvents(DiagramGraphicalViewer.java:256)
        at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramGraphicalViewer$2.run(DiagramGraphicalViewer.java:269)
        at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
        at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:185)
        at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:3933)
```
However, my local `CompaibilityPart` does not even have a line 473.

BeckmaR, 2019-02-18:
hey @ankls, I have a few questions:
* Do you have this button in the contextmenu on a transition?
[image]
* If yes, does it work?
* Can you save an error log immediately after you try to delete a transition and it fails? This would help us to pinpoint the actual exception that leads to the issue.

ankls, 2019-02-26:
I didn't have the issue again so far and the "Delete from Model" option is usable. But I'm very sure that the "Delete from Model" option was not there when I the issue was present.

I will go for a log as soon as I can reproduce the issue again.

BeckmaR, 2019-07-15:
Issue could not be reproduced again and is stale, closing. Please reopen if issue persists.

closed 2019-07-15 by BeckmaR