itemisCREATE/statecharts #2693: install problem

XiaoXiaoYi123, 2019-03-05:
Hello
 i have met this problem, I do not know what's wrong.Could you please help me?
ERROR: org.eclipse.pde.core code=0 Problems occurred while resolving the target contents
  at org.eclipse.oomph.util.OomphPlugin.coreException(OomphPlugin.java:280)
  at org.eclipse.oomph.util.pde.TargetPlatformUtil.activateTargetDefinition(TargetPlatformUtil.java:151)
  at org.eclipse.oomph.setup.pde.impl.TargetPlatformTaskImpl.perform(TargetPlatformTaskImpl.java:293)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.doPerformNeededSetupTasks(SetupTaskPerformer.java:3335)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.access$1(SetupTaskPerformer.java:3278)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil$1.run(SetupTaskPerformer.java:4514)
  at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2295)
  at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2322)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.performNeededSetupTasks(SetupTaskPerformer.java:4508)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.access$0(SetupTaskPerformer.java:4506)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performNeededSetupTasks(SetupTaskPerformer.java:3269)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performTriggeredSetupTasks(SetupTaskPerformer.java:3244)
  at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.perform(SetupTaskPerformer.java:3144)
  at org.eclipse.oomph.setup.ui.wizards.ProgressPage$9.run(ProgressPage.java:575)
  at org.eclipse.oomph.setup.ui.wizards.ProgressPage$11$1.run(ProgressPage.java:701)
  at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
  ERROR: org.eclipse.pde.core code=0 Problems loading repositories
    ERROR: org.eclipse.pde.core code=0 Problems loading repositories
      ERROR: org.eclipse.equinox.p2.transport.ecf code=1200 Artifact not found: http://download.eclipse.org/eclipse/updates/4.9/compositeContent.jar.
      java.io.FileNotFoundException: http://download.eclipse.org/eclipse/updates/4.9/compositeContent.jar
        at org.eclipse.equinox.internal.p2.transport.ecf.RepositoryStatusHelper.checkFileNotFound(RepositoryStatusHelper.java:297)
        at org.eclipse.equinox.internal.p2.transport.ecf.FileReader.checkException(FileReader.java:494)
        at org.eclipse.equinox.internal.p2.transport.ecf.FileReader.sendRetrieveRequest(FileReader.java:451)
        at org.eclipse.equinox.internal.p2.transport.ecf.FileReader.readInto(FileReader.java:378)
        at org.eclipse.equinox.internal.p2.transport.ecf.RepositoryTransport.download(RepositoryTransport.java:102)
        at org.eclipse.equinox.internal.p2.transport.ecf.RepositoryTransport.download(RepositoryTransport.java:158)
        at org.eclipse.equinox.internal.p2.repository.CacheManager.updateCache(CacheManager.java:405)
        at org.eclipse.equinox.internal.p2.repository.CacheManager.createCache(CacheManager.java:262)
        at org.eclipse.equinox.internal.p2.metadata.repository.CompositeMetadataRepositoryFactory.getLocalFile(CompositeMetadataRepositoryFactory.java:77)
        at org.eclipse.equinox.internal.p2.metadata.repository.CompositeMetadataRepositoryFactory.load(CompositeMetadataRepositoryFactory.java:100)
        at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.factoryLoad(MetadataRepositoryManager.java:63)
        at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.loadRepository(AbstractRepositoryManager.java:770)
        at org.eclipse.equinox.internal.p2.repository.helpers.AbstractRepositoryManager.loadRepository(AbstractRepositoryManager.java:671)
        at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:110)
        at org.eclipse.equinox.internal.p2.metadata.repository.MetadataRepositoryManager.loadRepository(MetadataRepositoryManager.java:105)
        at org.eclipse.pde.internal.core.target.P2TargetUtils.getQueryableMetadata(P2TargetUtils.java:910)
        at org.eclipse.pde.internal.core.target.IUBundleContainer.getRootIUs(IUBundleContainer.java:724)
        at org.eclipse.pde.internal.core.target.P2TargetUtils.getRootIUs(P2TargetUtils.java:1364)
        at org.eclipse.pde.internal.core.target.P2TargetUtils.resolveWithPlanner(P2TargetUtils.java:937)
        at org.eclipse.pde.internal.core.target.P2TargetUtils.synchronize(P2TargetUtils.java:762)
        at org.eclipse.pde.internal.core.target.TargetDefinition.resolve(TargetDefinition.java:371)
        at org.eclipse.oomph.util.pde.TargetPlatformUtil.activateTargetDefinition(TargetPlatformUtil.java:150)
        at org.eclipse.oomph.setup.pde.impl.TargetPlatformTaskImpl.perform(TargetPlatformTaskImpl.java:293)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.doPerformNeededSetupTasks(SetupTaskPerformer.java:3335)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.access$1(SetupTaskPerformer.java:3278)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil$1.run(SetupTaskPerformer.java:4514)
        at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2295)
        at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2322)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.performNeededSetupTasks(SetupTaskPerformer.java:4508)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer$WorkspaceUtil.access$0(SetupTaskPerformer.java:4506)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performNeededSetupTasks(SetupTaskPerformer.java:3269)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.performTriggeredSetupTasks(SetupTaskPerformer.java:3244)
        at org.eclipse.oomph.setup.internal.core.SetupTaskPerformer.perform(SetupTaskPerformer.java:3144)
        at org.eclipse.oomph.setup.ui.wizards.ProgressPage$9.run(ProgressPage.java:575)
        at org.eclipse.oomph.setup.ui.wizards.ProgressPage$11$1.run(ProgressPage.java:701)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
    ERROR: org.eclipse.pde.core code=0 Problems loading repositories
      ERROR: org.eclipse.pde.core code=0 Unable to locate installable unit com.google.inject 3.0.0.v201605172100
      ERROR: org.eclipse.pde.core code=0 Unable to locate installable unit com.google.inject.multibindings 3.0.0.v201605172100
      ERROR: org.eclipse.pde.core code=0 Unable to locate installable unit org.apache.commons.io 2.2.0.v201405211200

BeckmaR, 2019-03-05:
Please try again, eclipse mirrors are sometimes down.

BeckmaR, 2019-03-12:
Closing this because of inactivity

closed 2019-03-12 by BeckmaR