itemisCREATE/statecharts #2706: profile

XiaoXiaoYi123, 2019-03-13:
Hey,

 I just want to do some work on the yankindu.

 What i want to do is to generate a status diagram from an excel table.
My idea is to first specify the contents of the excel table, then convert excel to XML format, and finally convert the XML format to SCT file format. I intend to do this using an eclipse-based EMF (ecore) framework, followed by an ATL model transformation technique. I don't know if this idea works, or if you have a more reliable idea. I am looking forward to your help and appreciate it very much.

BeckmaR, 2019-03-13:
Hey @XiaoXiaoYi123, I'm sorry but I fear that what you want to achieve is nearly impossible. While the route you want to take would probably work on a technical level, there would still be major issues. The reason for this is that the *.sct files also contain information about the graphical layout of the statechart, which you would have to generate (and create) as well, which contains for example transition routing and so on. You could take a look at existing *.sct-files, you will see that they basically contain two sections - the information about the states and their logical connections, and another one with graphical information.

I would be very pleased to answer your further questions on our [user forums](https://groups.google.com/forum/#!forum/yakindu-user). This issue tracker is better suited for actual issues, for example bugs and feature requests.

closed 2019-03-13 by BeckmaR

XiaoXiaoYi123, 2019-03-14:
Hello, I am very interested in this tool.

I've been working on the.sct file and I've learned that it's an xmi file and I'd like to know how this.sct file is generated.

I see that the.sct file has two pieces of information, one about the state of the state diagram and the migration between states, and the other about the UI information for each node. In addition, I saw the sgraph.ecore and stext.ecore files in the source code, and I found that sgraph.ecore is the structure part of the description of the state diagram, and stext.ecore is the behavior information of the description of the state diagram.

I have the following questions:

1. Are the states in the. SCT file and the migration information between them generated by the sgraph.ecore and stext.ecore files?

2. If 1 is true, how to generate UI information of each node? Through the GEF framework?

3. If I want to generate a. SCT file from excel, I need to create an excel ecore model. In addition, regarding the UI information of nodes, could you give a fixed value first, then open it with yakindu, and then modify the corresponding information?

My question is a little bit doha, I really need your help, if you can reply to me, I will be very grateful. Looking forward to your reply.

If I say something wrong, I hope you can point it out. 

I wish good