itemisCREATE/statecharts #2712: Error in C# code generator

stuartdl, 2019-03-21:
Yakindu code version is 3.5.2

When using the attached state chart project, when it generates the C# code, the code contains
[exampleStateChart.zip](https://github.com/Yakindu/statecharts/files/2994383/exampleStateChart.zip)
 references to a function called react
e.g.
```
		bool react() {
			return false;
		}
```
		
the problem is, in the generated code, sometimes the function is called like
`react()==false`
and sometimes it is called like
`react(false)==false;`

Either the code generator is adding a "false" as a parameter when it shouldn't 
OR
the code generator isn't creating an overloaded version of the react() method that takes a bool as a parameter.

tkutz, 2019-03-27:
Thanks for the bug report!

This will be fixed in version 3.5.3 coming out end of this week.

tkutz, 2019-03-27:
Fixed with https://github.com/Yakindu/statecharts/pull/2723

closed 2019-03-27 by tkutz

stuartdl, 2019-04-01:
Thankyou Thomas,
Much appreciated

PR #2723 (2019-03-27): Fixes in C# code generator
Fixes #2712 
Fixes #2721