itemisCREATE/statecharts #2718: No error will be reported when In-Event is raised

DefCon-CC, 2019-03-25:
We observed the Problem in Yakindu SCT 3.4.1. that an In-Event can be raised. Which is from our point of view an error. Example:

[image]

[image]

The code is generated without any further warning or error.

BeckmaR, 2019-03-26:
Agree. At least a warning should be created, and we should deprecate this "feature" altogether.

terfloth, 2020-11-26:
With release 4.x raising in events by the state machine itself is clearly defined using event buffering which is enabled by default. If it is disabled we should add warnings.