itemisCREATE/statecharts #2725: C# Code Generator misplaces a function definition in the statemachine interfaces files

stuartdl, 2019-03-28:
Yakindu code version is 3.5.2 with the C# code generator fixes from #2723

When using the attached state chart project, when it generates the C# code, the ITestStatemachine.cs file has the following line added to the bottom of the file (which stops compilation)
`	void setInternalOperationCallback(InternalOperationCallback operationCallback);`
I suspect it should be included in one of the generated interfaces.
[ExampleCode.zip](https://github.com/Yakindu/statecharts/files/3015862/ExampleCode.zip)

tkutz, 2019-03-28:
Yes, the setter schould be placed inside the statemachine's interface. This will be fixed in version 3.5.3.

closed 2019-03-28 by tkutz

tkutz, 2019-03-29:
I just want to inform you that version 3.5.3 is out now. You can download it here: https://info.itemis.com/state-machine/download-yakindu-statechart-tools