itemisCREATE/statecharts #2728: Not declaring always trigger is implicit when guards are used

rherrmannr, 2019-03-28:
The behavoir of not using always triggers is not consistent.
Always trigger will be used implicit, if guards are used.

In my opinion, there should be a warning caused by the missing trigger. StateC should not be entered.

I'm not sure if we should change this, because this would have an impact on the execution behavior of the simulation and the generated code.

[image]

BeckmaR, 2019-05-24:
StateC is just taken here because transitions without trigger are never taken, so transition 2 is taken instead - as the warning says. I don't get the issue...

andreasmuelder, 2019-11-08:
As the topic says, when a guard is defined an implicit always trigger is added. This is the desired behaviour.

closed 2019-11-08 by andreasmuelder