itemisCREATE/statecharts #2731: C# code generator doesn't support event driven mode..can i help?

stuartdl, 2019-04-03:
The C# code generator only generates cycle driven code, this causes all internal events to be cleared at the end of RunCycle() even when Event driven mode is defined.
I have looked at the generated code for the C# and the C state machines and it appears that an event queue needs to be implemented in C# and then integrated into the code generator.
Is there any C# code i can write (e.g. an event queue) and submit that will help get this feature implemented?

tkutz, 2019-04-04:
That's correct, we have enhanced the C, C++ and Java code generators with event queues, but not yet the C# generator.

Basically, there are two event queues, (1) the internal event queue (for internal events), and (2) the in-event queue (for incoming events). While the internal event queue is always generated, the in-event queue is optional (can be enabled in sgen model). Events in the internal event queue are processed with higher priority than those in the in-event queue.

Currently we are refactoring our code generators for our 4.0 version, in which the C# generator will be also updated. However, this will not be finished until end of Q3 2019.

That said, we would really appreciate it if you could contribute some C# implementation of this feature. It might make sense to have a look at the generated Java code.

tkutz, 2019-04-08:
Thanks a lot! We will take a look and come back to you.

tkutz, 2019-04-12:
@stuartdl, we will start with the implementation in May. However, since we are currently focused on restructuring our code base, we won't be able to test this feature profoundly. So it would be best to do some iterations together. We will let you know once we have a first running prototype.

tkutz, 2019-05-14:
@stuartdl, we added support for event driven statecharts to the C# code generator. You can download the updated version (3.5.4) from the labs update site: http://updates.yakindu.com/statecharts/labs

In case of any problems, please reopen this issue, or create a new one.

PR #2746 (2019-05-10): C# Internal Event Queue
Fixes #2731: Adds internal event queue to C# when event driven semantics are used.

@tangele please try this out and review
closed 2019-05-14 by tkutz