itemisCREATE/statecharts #2735: Floating License: Crash when license is already in use

DanielBernhard, 2019-04-03:
Using eclipse oxgen with yakindu 3.3 (as available through marketplace) I get strange behavior when our floating license is not available:
1. The first time the license is not available, everything works as intended (diagrams are readonly, license is shown as "missing").
2. When restarting eclipse now (and the license is still missing) I cannot view the diagrams anymore (empty page is displayed).
3. When going to the preferences "YAKINDU Licenses" an error message pops up: "The currently displayed page contains invalid values".
4. To actually see the page I need to switch pages a couple of times. All boxes in the dialog are empty now (except for Port which is set to 6200).
5. When I re-enter our license server details and press "Apply and Close" I get the an exception dialog: "Could not initialize class com.yakindu.licmgmt.IEclipseLicenseService" (see attached log file)
6. The only approach I currently now to "fix" the problem is to switch workspaces. The first time after switching workspace everything works fine again.
[Eclipse.log](https://github.com/Yakindu/statecharts/files/3038716/Eclipse.log)
[EclipseConfig.txt](https://github.com/Yakindu/statecharts/files/3038717/EclipseConfig.txt)

tkutz, 2019-04-03:
It is hard to say what went wrong in your case. Your eclipse config says that you have SCT 3.5.3 installed, so I guess you installed the (outdated) 3.3.0 from marketplace and then did an update to 3.5.3. Somewhere on that route the license relevant plugins got confused/corrupted.

You can try to start your Eclipse from command line with "-clean" option, this should clean up some caches.

If that does not work, I suggest to install a fresh 3.5.3 version of yakindu.

tkutz, 2019-04-15:
@DanielBernhard Did you manage to setup your SCT properly or do you need more help?

DanielBernhard, 2019-05-28:
Sorry for the delay in answering: The problem persists (current not very urgent as I'm working on a different part of the project, but ...)
I actually deleted my existing workspace, reinstalled eclipse from scratch. No help... The first time works, later it doesn't...

andreasmuelder, 2019-06-14:
@DanielBernhard thanks for the reply. We found an issue an this will be fixed with the upcoming 3.5.5 version we plan to release next week.

andreasmuelder, 2019-06-18:
we released the bugfix today - please reopen if the error still exists

closed 2019-06-18 by andreasmuelder

DanielBernhard, 2019-06-19:
Looks good so far. Thanks a lot!