itemisCREATE/statecharts #2742: Error message is weak for "exit is not connected to any outgoing transitions"

DefCon-CC, 2019-05-06:
In Yakindu SCT 3.4.1 we got the following error message when generating the state machine

```
Generating 'Example' to target project 'Appl' ...
The referenced model (Example) contains errors and could not be generated:
This exit is not connected to any outgoing transitions.
Example done.
```

But for large state machines (with many sub-diagramms) it is very unprecise where the error "This exit is not connected to any outgoing transitions." occured. A remark in which region, sub-diagram, ... the error occured would be helpful!

andreasmuelder, 2019-05-06:
If the model is faulty, you should get a problem marker in the Problems View. You can jump to the location with a double click. Is this not sufficient?

DefCon-CC, 2019-05-06:
> If the model is faulty, you should get a problem marker in the Problems View. 

I do not see any problem marker in the Problems View. I see also in the model no error

andreasmuelder, 2019-05-06:
It should look like this.
[image]

Maybe "Build automatically" is disabled in your workspace? Do you see an error marker after project / build?

DefCon-CC, 2019-05-06:
Ahh, the problem was introduced by second *.sct file in the same folder with the same StaeChart Name I keept as backup. Ticket can be closed!

closed 2019-05-06 by DefCon-CC