itemisCREATE/statecharts #2799: Install problems-require your help

XiaoXiaoYi123, 2019-09-05:
hello, when i installed  Yakiindu from source code, there existed no problems, but when  I run Yakindu from source code, it showed "Installation is currupt . License is missing. - editor is in readonly mode"
[image]

tkutz, 2019-09-05:
This is actually changed on purpose in version 3.5.6 to avoid usage of Yakindu without having a valid license. If you want to work on the code for your private project you can always switch to the 3.5.5 tag.

XiaoXiaoYi123, 2019-09-05:
There is another small request. I don't know where to get the 3.5.5 version. Thank you very much for your help

XiaoXiaoYi123, 2019-09-05:
Hello, I installed it according to your installation tutorial. If I want to install version 3.5.5, should I reinstall it or modify the configuration? In addition, where should I find the installation package if I want to reinstall it？

tkutz, 2019-09-05:
You just need to checkout the corresponding git tag `release_3.5.5`.

XiaoXiaoYi123, 2019-09-05:
Ok，thanks for your help

XiaoXiaoYi123, 2019-09-05:
I'm sorry to bother you again. When I have installed the Yakindu release 3.5.5 from source code. The problem has already existed. Does that mean we can't operate on the GUI without a license？I also tried to install Yakindu from eclipse Marketplace, also has a problem same as this  one.

tkutz, 2019-09-06:
What do you mean with "installed from source code?".

XiaoXiaoYi123, 2019-09-06:
I followed this instruction to install the Yakindu release 3.5.5:
[image]
 But I found that the link 
"
[image]
"
is the same as the release 3.5.6.
I guess that this may cause problems when I met in release 3.5.6.(the editor is only read)
I don't know how to install the release 3.5.5, so I really need your help.
Thank you sincerely!

tkutz, 2019-09-09:
When used `StatechartTools.setup` the installer will download the source code into a local git repository. In this git repository you can switch branches/tags. Default is the current `master`. You would need to switch to the `3.5.5` tag. You can do this in your eclipse or command line or whatever tool you use to manage your git repos.

XiaoXiaoYi123, 2019-09-09:
I understand what you mean if used 
[image]
the source code in GitHub will be downloaded locally，like this 
[image]
 and then I can modify the version of Yakindu in this git repository.
You said that I can modify the Yakindu tool version in eclipse. Do I need to replace the files in the git repository with the 3.3.5 version? I am very sorry that I did not understand your meaning clearly. I am looking forward to your reply. Thank you very much

andreasmuelder, 2019-11-08:
closing as orphaned

closed 2019-11-08 by andreasmuelder

kubsztal, 2020-09-03:
> If you want to work on the code for your private project you can always switch to the 3.5.5 tag.

How this restriction relates to the EPL license which allows to make any modification to the code anyway, thus one can simply revert this change in order to use the software above the version 3.5.5?