itemisCREATE/statecharts #2872: SCT Fails to open

larsrengersen, 2019-11-09:
Just downloaded the latest version of Statecharts.
Unfortunately the App does not open and only gives an alert

Failed to find a Main Class in "/Applications/YAKINDU SCT.app/Contents/MacOS//../Eclipse/plugins/org.eclipse.equinox.launcher_1.5.100.v20180827-1352.jar".

[image]

Already updated Java to the latest version (Java 8 build 231) but that did not make any difference. I'm using Mac OS 10.14.6
Any help on how to open the App is appreciated.

larsrengersen, 2019-11-10:
After installing JDK 13 (not only JRE), it now opens.

larsrengersen, 2019-11-11:
Well, it's not going very smooth for me.
I am trying to install/integrate Sloeber since in the end I want to generate Arduino code.
I am getting the error 
The installpath can not contain spaces /Applications/YAKINDU SCT.app/Contents/Eclipse/arduinoPlugin
Sloeber might still function but if you get strange results you know where to look.
Do not create an issue if you see this!!!
The last line is not very encouraging since after reading many pages I still do not know how to get it working. This page http://eclipse.baeyens.it/installAdvice.shtml says "Take the product and not the plugin." and I did use the Plugin, but the product gives another error 
"Cannot complete the install because one or more required items could not be found.
  Software being installed: Sloeber, the Eclipse Arduino IDE 4.3.1.201810101819 (io.sloeber.product 4.3.1.201810101819)
  Missing requirement: GNU MCU C/C++ OpenOCD Debugging 4.3.3.201809300659 (ilg.gnumcueclipse.debug.gdbjtag.openocd 4.3.3.201809300659) requires 'osgi.bundle; ilg.gnumcueclipse.debug.gdbjtag 0.0.0' but it could not be found"
Hope someone can point me towards the right direction to get it working.

BeckmaR, 2019-11-11:
Hi @larsrengersen, thank you for trying out YAKINDU SCT! I'm happy to hear you could solve your Java issues. For some reason quite a few users had issues with Java in the recent weeks.

The sloeber plugin is a bit finnicky at times. The plugin they are missing means they have some error in their setup. I think you need to install this:
[GNU MCU Eclipse](https://marketplace.eclipse.org/node/2593184)
If you want to develop for Arduino, there are other plugins you can try:
[Eclipse C/C++ for Arduino](https://marketplace.eclipse.org/node/2562013)

Hope this helps!

larsrengersen, 2019-11-11:
Thanks! I have explored the other Arduino plugin and unfortunately that does not support the Teensy 3.6 board I am using. Have given Sloeber another try and when choosing the "plugin" version I can open SCT but do not manage to find out how to write to the Teensy. Then I am running into new errors. Think I am going to leave it as is for now and continue writing state machines by hand in the Arduino IDE. For now the learning curve for getting Yakindu SCT up and running is too steep for me. Still like the concept/approach though so perhaps I will give it another try in the future.
Thanks again for your reply and hints.

BeckmaR, 2019-11-11:
I'm sorry, I can't really help you much with the sloeber plugin. Maybe you can get some support from its author.

There's another option to combine SCT and the regular Arduino IDE that I quite like myself, using Arduino libraries.
If you navigate to your Arduino sketch directory, there will be a folder named libraries - for me that's `/Users/rbeckmann/Documents/Arduino/libraries`. With this pathname, go back to SCT (without sloeber or anything) and create a new C/C++ project, but choose the library folder as the project location.
[image]
Continue to create the project, statechart etc and just generate code. After that, you should be able to add the project as a library from the arduino IDE and use the generated state machine. If the library is not detected instantly open the library manager once, I think this rescans the path or something.

closed 2019-12-09 by andreasmuelder