itemisCREATE/statecharts #2903: Migrate to newer googletest verion

andreasmuelder, 2020-01-10:
We have to update the googletest framework since GCC > 7.1 throws a warning with the current tests:
1.8.1 + 1.10.0 do not compile for C++:

Compilation failed (exit status 1):
/home/travis/build/Yakindu/statecharts/googletest-release-1.8.1/googlemock/build/libgmock_main.a(gmock_main.cc.o): In function `main':
gmock_main.cc:(.text+0x31): undefined reference to `testing::InitGoogleMock(int*, char**)'
collect2: error: ld returned 1 exit status
	at org.yakindu.sct.generator.cpp.test.OperationsTest.setUp(OperationsTest.java:33)

rherrmannr, 2020-02-10:
There was a bug report: https://github.com/google/googletest/issues/1094
But seems like there is no fix.

rherrmannr, 2020-02-10:
Actually, we don't use anything from gmock. We can simply remove the include.

PR #2975 (2020-02-10): Issue 2903 remove gmock
fix #2903 

The problem was a warning in the gmock header compiled with GCC >= 7.1

BUT we don't use anything of gmock. There WAS the plan to use the mocking features of gmock, but they've never been used! So, we simply can remove the gmock include, which should be a simple workaround for this problem.

Tests have been regenerated with https://github.com/Yakindu/sctpro/pull/2230
closed 2020-02-10 by andreasmuelder