itemisCREATE/statecharts #2942: SCT editor doesn't load on Linux / Eclipse 2019-12

bbpaulvanharen, 2020-01-27:
I've a fresh install of Eclipse 2019-12 on Linux, and loaded SCT through Eclipse Marketplace. As shown, Eclipse doesn't render the contents of the state diagram correctly. The same workspace opened on Eclipse Photon on the same Linux computer renders the state diagram as it should (second image).
[image]
[image]

andreasmuelder, 2020-02-06:
@rherrmannr can you have a look at this?

rherrmannr, 2020-02-06:
I can reproduce this. I will have a closer look.

andreasmuelder, 2020-02-11:
Seems to be a regression with GTK

org.eclipse.swt.SWTException: Graphic is disposed
	at org.eclipse.swt.SWT.error(SWT.java:4720)
	at org.eclipse.swt.SWT.error(SWT.java:4635)
	at org.eclipse.swt.SWT.error(SWT.java:4606)
	at org.eclipse.swt.graphics.Color.getRed(Color.java:303)
	at org.eclipse.swt.graphics.Color.equals(Color.java:233)
	at org.eclipse.draw2d.SWTGraphics.checkPaint(SWTGraphics.java:303)
	at org.eclipse.draw2d.SWTGraphics.drawRectangle(SWTGraphics.java:503)
	at org.eclipse.draw2d.Graphics.drawRectangle(Graphics.java:351)
	at org.eclipse.draw2d.LineBorder.paint(LineBorder.java:143)
	at org.eclipse.draw2d.Figure.paintBorder(Figure.java:1137)
	at org.eclipse.draw2d.Figure.paint(Figure.java:1121)
	at org.eclipse.draw2d.Figure.paintChildren(Figure.java:1170)
	at org.eclipse.draw2d.Figure.paintClientArea(Figure.java:1205)
	at org.eclipse.draw2d.Figure.paint(Figure.java:1120)
	at org.eclipse.draw2d.Figure.paintChildren(Figure.java:1170)
	at org.eclipse.draw2d.Figure.paintClientArea(Figure.java:1205)
	at org.eclipse.draw2d.Figure.paint(Figure.java:1120)
	at org.eclipse.draw2d.Figure.paintChildren(Figure.java:1170)
	at org.eclipse.draw2d.Figure.paintClientArea(Figure.java:1200)
	at org.eclipse.draw2d.Viewport.paintClientArea(Viewport.java:160)
	at org.eclipse.draw2d.Figure.paint(Figure.java:1120)
	at org.eclipse.draw2d.Figure.paintChildren(Figure.java:1170)
	at org.eclipse.draw2d.Figure.paintClientArea(Figure.java:1205)
	at org.eclipse.draw2d.Figure.paint(Figure.java:1120)
	at org.eclipse.draw2d.DeferredUpdateManager.paint(DeferredUpdateManager.java:168)
	at org.eclipse.draw2d.LightweightSystem.paint(LightweightSystem.java:203)
	at org.eclipse.draw2d.LightweightSystem$2.handleEvent(LightweightSystem.java:110)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5676)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1423)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1449)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1432)
	at org.eclipse.swt.widgets.Control.gtk_draw(Control.java:3925)
	at org.eclipse.swt.widgets.Scrollable.gtk_draw(Scrollable.java:342)
	at org.eclipse.swt.widgets.Composite.gtk_draw(Composite.java:455)
	at org.eclipse.swt.widgets.Canvas.gtk_draw(Canvas.java:181)
	at org.eclipse.swt.widgets.Widget.windowProc(Widget.java:2243)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:6826)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5917)
	at org.eclipse.swt.internal.gtk.GTK._gtk_main_do_event(Native Method)
	at org.eclipse.swt.internal.gtk.GTK.gtk_main_do_event(GTK.java:4168)
	at org.eclipse.swt.widgets.Display.eventProc(Display.java:1480)
	at org.eclipse.swt.internal.gtk.OS._g_main_context_iteration(Native Method)
	at org.eclipse.swt.internal.gtk.OS.g_main_context_iteration(OS.java:1604)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4427)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1049)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:660)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:154)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:657)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:594)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1465)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1438)
[... 34 lines cut ...]
	at org.eclipse.swt.widgets.Control.gtk_draw(Control.java:3925)
	at org.eclipse.swt.widgets.Scrollable.gtk_draw(Scrollable.java:342)
	at org.eclipse.swt.widgets.Composite.gtk_draw(Composite.java:455)
	at org.eclipse.swt.widgets.Canvas.gtk_draw(Canvas.java:181)
	at org.eclipse.swt.widgets.Widget.windowProc(Widget.java:2243)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:6826)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5917)
	at org.eclipse.swt.internal.gtk.GTK._gtk_main_do_event(Native Method)
	at org.eclipse.swt.internal.gtk.GTK.gtk_main_do_event(GTK.java:4168)
	at org.eclipse.swt.widgets.Display.eventProc(Display.java:1480)
	at org.eclipse.swt.internal.gtk.OS._g_main_context_iteration(Native Method)
[... cut ...]
	at org.eclipse.jface.viewers.StructuredViewer$1.run(StructuredViewer.java:777)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.ui.internal.JFaceUtil.lambda$0(JFaceUtil.java:47)
	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:174)
	at org.eclipse.jface.viewers.StructuredViewer.fireDoubleClick(StructuredViewer.java:774)
	at org.eclipse.jface.viewers.AbstractTreeViewer.handleDoubleSelect(AbstractTreeViewer.java:1518)
	at org.eclipse.jface.viewers.StructuredViewer$4.widgetDefaultSelected(StructuredViewer.java:1205)

rherrmannr, 2020-02-20:
@bbpaulvanharen we could figure out the problem. It will be fixed with our next release (3.5.10).

Please re-open this issue, if it still exist.

closed 2020-02-20 by rherrmannr