itemisCREATE/statecharts #3004: Missig dispatch for org.yakindu.sct.model.sgraph.impl.SynchronizationImpl

pavelpa, 2020-02-20:
I'm experiencing an issue to generate sxml file for a model containing a simple orthogonal states.

Using:
Yakindu SCT Professional Edition version 3.5.3
Created model: SCXML Domain
YAKINDU SCT SCXML Generator

Issue 1: Generated scxml file sets initial state to "sync0", which is not defined.
Issue 2: Generated scxml file contains: "Missig dispatch for org.yakindu.sct.model.sgraph.impl.SynchronizationImpl@1f9e36a5 (name: null)"

`<?xml version="1.0" encoding="UTF-8"?>
<scxml xmlns="http://www.w3.org/2005/07/scxml" version="1.0" datamodel="ecmascript" name="default" initial="sync0">
	<state id="main_region">
		**Missig dispatch for org.yakindu.sct.model.sgraph.impl.SynchronizationImpl@1f9e36a5 (name: null)**
		<parallel id="OrthoTest">
			<state id="r1">
				<initial>
					<transition target="StateA" type="internal" />
				</initial>
				<state id="StateA">
				</state>
			</state>
			<state id="r2">
				<initial>
					<transition target="StateB" type="internal" />
				</initial>
				<state id="StateB">
				</state>
			</state>
			<transition   target="default.final_0">
			</transition>
		</parallel>
		<final id="default.final_0"/>	
	</state>
</scxml>`

andreasmuelder, 2020-02-20:
Thanks @pavelpa for reporting, I will take a look. 
Did you try with latest release 3.5.9?

pavelpa, 2020-02-25:
I'm not able to update to latest release:

Some sites could not be found.  See the error log for more detail.
Unknown Host: http://updates.yakindu.com/statecharts/releases/content.xml
updates.yakindu.com
Unknown Host: http://updates.yakindu.com/statecharts/labs/content.xml
updates.yakindu.com
Unknown Host: http://download.eclipse.org/releases/2018-09/content.xml
download.eclipse.org

pavelpa, 2020-03-24:
"Missing dispatch ...." is not really an issue here, even it breaks xml syntax. QtScxmlStateMachine simply ignores it. The issue here is that generated scxml file contains references to states, that do not exist.

andreasmuelder, 2020-03-25:
This problem occures when a synchronization for fork semantics is used.
Join semantics works:
![join_scxml_sync](https://user-images.githubusercontent.com/6338708/77553636-
56a4d380-6eb5-11ea-9c5a-9e690540fd7b.png)

Fork semantic generates the missing dispatch:
[image]

andreasmuelder, 2020-04-01:
Fixed for upcoming 3.5.11 release

closed 2020-04-01 by andreasmuelder