itemisCREATE/statecharts #3010: Dead transition not marked as such

Mazesch, 2020-02-20:
If there are two timed transitions with different times, the longer one will never be taken, since the first one is always faster.
But the slower transition is not marked as `dead`.

[image]

tkutz, 2020-02-20:
Actually, this is even more tricky. Consider the case in which the transition prios are swapped (the longer time trigger has prio 1); then in the CycleBased case, with cycles higher than 100ms, both triggers would be valid in the same RTC step, and because of the prio, the 101ms trigger would be taken instead of the 100ms one.

terfloth, 2020-11-26:
Requires model checking ...

closed 2020-11-26 by terfloth