itemisCREATE/statecharts #3018: New SCT model doesn't show in editor

bbpaulvanharen, 2020-02-27:
When creating a new statechart model, a non-functioning screen is editor window. The editor window doesn't display the filename, and has an inactive canvas, see screenshot. 
In the project explorer, the properties of the newly created statechart model are available. 

As a test, I've copied the contents an existing model into the new model (file based copy on file system level). In that case, the editor is opening correctly and showing contents as usual.

Please note, I'm using Yakindu SCT v 3.5.10 on Eclipse 2019-12 on Kubuntu 18.04 LTS.

Attachments:
Screenshot.png showing the inactive editor
[image]
* (https://github.com/Yakindu/statecharts/files/4260735/eclipse.log) the logfile produced by the eclipse session
* (https://github.com/Yakindu/statecharts/files/4260738/default.sct.txt)
 the newly created model

rherrmannr, 2020-02-28:
How did you get the 3.5.10 version? Via update, update site, fresh install?
Currently, I can not reproduce the error. I've used an empty 2019-12 eclipse and installed SCT via update site on Ubuntu 18.04. 

New models work and I also can open your uploaded model.

bbpaulvanharen, 2020-03-02:
I've a fresh install of eclipse 2019-12. I installed SCT from the eclipse marketplace, i.e. the demo version. Immediately thereafter, I installed SCT from the "Install new software"  menu option, using the link that came along with the installation from the marketplace: http://updates.yakindu.com/statecharts/releases/

rherrmannr, 2020-03-02:
I can not reproduce this.

So, here is what I did:
- using the oomph installer and installed a clean 2019-12 
[image]
- Manually installed the market place
[image]
- Installed SCT via Marketplace
- Created a new model (working)
- Using "Install new Software" and added the update site to install YAKINDU Statechart Tools Standard Edition
- Still, the old model does work
- Added a new model that worked fine

Did I miss anything? Could you try out this solution, please?

bbpaulvanharen, 2020-03-04:
Hi,
Thank you for trying to reproduce my steps. I think you copied my scenario just fine. The issue remains; it works in your environment and not in mine. 
To be sure, I reinstalled a fresh Eclipse CDT through Oomph, version 2019-12, installed SCT from the marketplace and then again installed new software. In a new workspace, using the fresh install, I was able to reproduce the problem just fine.

I ran `eclipse > eclipse.log 2&>1` to grab all debug and logging output. Please find that
[eclipse.log](https://github.com/Yakindu/statecharts/files/4288067/eclipse.log) attached. I hope it contains some useful clues.

andreasmuelder, 2020-03-16:
Hi @bbpaulvanharen,
thanks, I checked the log, it seems that your installation does not contain the license management feature and thus some classes could not be loaded. Since I couldn't reproduce it too I will close this issue - it seems to be a specific problem with your environment. As a workaround, you can install the Licence Management feature from our provided update site.

closed 2020-03-16 by andreasmuelder