itemisCREATE/statecharts #3021: UI Freeze and lost changes with RC4

rherrmannr, 2020-02-28:
After 10 minutes of modeling, I had several errors of this kind:
```
java.lang.IllegalArgumentException: Argument not valid
	at org.eclipse.swt.SWT.error(SWT.java:4701)
	at org.eclipse.swt.SWT.error(SWT.java:4635)
	at org.eclipse.swt.SWT.error(SWT.java:4606)
	at org.eclipse.swt.graphics.TextLayout.setFont(TextLayout.java:1805)
	at org.eclipse.swt.custom.StyledTextRenderer.getTextLayout(StyledTextRenderer.java:1173)
	at org.eclipse.swt.custom.StyledTextRenderer.getTextLayout(StyledTextRenderer.java:888)
	at org.eclipse.swt.custom.StyledText.getPointAtOffset(StyledText.java:5560)
	at org.eclipse.swt.custom.StyledText.setCaretLocation(StyledText.java:8710)
	at org.eclipse.swt.custom.StyledText.handleResize(StyledText.java:6319)
	at org.eclipse.swt.custom.StyledText.lambda$1(StyledText.java:5801)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5676)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1423)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1449)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1428)
	at org.eclipse.swt.widgets.Control.setBounds(Control.java:1288)
	at org.eclipse.swt.widgets.Composite.setBounds(Composite.java:1599)
	at org.eclipse.swt.widgets.Canvas.setBounds(Canvas.java:461)
	at org.eclipse.swt.widgets.Control.setBounds(Control.java:1068)
	at org.yakindu.base.xtext.utils.gmf.directedit.XtextDirectEditManager$2.relocate(XtextDirectEditManager.java:256)
	at org.yakindu.base.xtext.utils.gmf.directedit.DirectEditManagerEx.placeCellEditor(DirectEditManagerEx.java:342)
	at org.yakindu.base.xtext.utils.gmf.directedit.DirectEditManagerEx.access$1(DirectEditManagerEx.java:341)
	at org.yakindu.base.xtext.utils.gmf.directedit.DirectEditManagerEx$1.ancestorMoved(DirectEditManagerEx.java:264)
	at org.eclipse.draw2d.AncestorHelper.fireAncestorMoved(AncestorHelper.java:104)
	at org.eclipse.draw2d.AncestorHelper.figureMoved(AncestorHelper.java:91)
	at org.eclipse.draw2d.Figure.fireFigureMoved(Figure.java:496)
	at org.eclipse.draw2d.PolylineConnection.layout(PolylineConnection.java:188)
	at org.eclipse.draw2d.Figure.validate(Figure.java:1899)
	at org.eclipse.draw2d.Figure.validate(Figure.java:1901)
	at org.eclipse.draw2d.Figure.validate(Figure.java:1901)
	at org.eclipse.draw2d.Figure.validate(Figure.java:1901)
	at org.eclipse.draw2d.Figure.validate(Figure.java:1901)
	at org.eclipse.draw2d.FreeformViewport$FreeformViewportLayout.calculatePreferredSize(FreeformViewport.java:25)
	at org.eclipse.draw2d.AbstractLayout.getPreferredSize(AbstractLayout.java:110)
	at org.eclipse.draw2d.AbstractHintLayout.getPreferredSize(AbstractHintLayout.java:90)
	at org.eclipse.draw2d.Figure.getPreferredSize(Figure.java:810)
	at org.eclipse.draw2d.ScrollPaneSolver.solve(ScrollPaneSolver.java:82)
	at org.eclipse.draw2d.FigureCanvas.layoutViewport(FigureCanvas.java:325)
	at org.eclipse.draw2d.FigureCanvas.access$4(FigureCanvas.java:323)
	at org.eclipse.draw2d.FigureCanvas$3.notifyValidating(FigureCanvas.java:292)
	at org.eclipse.draw2d.UpdateManager.fireValidating(UpdateManager.java:143)
	at org.eclipse.draw2d.DeferredUpdateManager.performValidation(DeferredUpdateManager.java:217)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramGraphicalViewer$ToggleUpdateManager.performValidation(DiagramGraphicalViewer.java:116)
	at org.eclipse.draw2d.DeferredUpdateManager.performUpdate(DeferredUpdateManager.java:193)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramGraphicalViewer$ToggleUpdateManager.performUpdate(DiagramGraphicalViewer.java:106)
	at org.eclipse.draw2d.DeferredUpdateManager$UpdateRequest.run(DeferredUpdateManager.java:44)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:185)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4910)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4432)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1049)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:660)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:559)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:154)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:150)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:657)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:594)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1465)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1438)
```

andreasmuelder, 2020-03-10:
The Deadlock * should * be fixed with https://github.com/Yakindu/statecharts/pull/3034

andreasmuelder, 2020-03-10:
The exception seems to be a linux problem (again) any steps to reproduce?

rherrmannr, 2020-03-10:
Nope, as I said. Used the tool for ~10 minutes and then it freezes. It just happened one time.

andreasmuelder, 2020-03-16:
if this is ont fixed, reopen with steps to reproduce

closed 2020-03-16 by andreasmuelder