itemisCREATE/statecharts #3029: Generate a function to get the state name when tracing feature is enabled

jfgo20, 2020-03-05:
It would be nice if a function would be generated to get the state name when the tracing feature is enabled. Right now, I need to have this kind of function in my trace observer:

```
const char* TraceObserverChartX::getStateName(ChartX::ChartXStates state) const {
  switch (state) {
    case ChartX::ChartX_last_state:
      return "Invalid";
    case ChartX::Char_X_ENTRY:
      return "Entry";
    case ChartX::Char_X_DOTHIS:
      return "Do this";
    case ChartX::Char_X__final_:
      return "Final";
    default:
      return "Unknown";
  }
}
```

So, right now, when I create a new state, I always need to remember to add it here.

jfgo20, 2020-03-05:
Also, if it was possible to reorder the items in the state enumeration, it would be easier to maintain without a generated function. For example, right now, the final state can be located anywhere in the enumeration, depending on when it was added.

rherrmannr, 2020-03-06:
We're currently working on an extended version of tracing, called YET ([Yakindu Execution Tracing](https://blogs.itemis.com/en/embedded-testing-and-debugging-with-state-machines)). This extension will provide an array with all the existing state names. 

This is still work in progress, but that could be the result:
```
static sc_string state_identifier[] = {
	"<nostate>",
	"main_region.Start",
	"main_region.TicToc",
	"main_region.TicToc._region0.Tic",
	"main_region.TicToc._region0.Toc"
};
```

This might be very useful for your case.

jfgo20, 2020-03-06:
Nice! Is there any version available right now?

rherrmannr, 2020-03-06:
The first prototype, which can be released, is in progress - probably within the next 1-2 weeks. I will provide a plugin as soon as possible.

But note: It's completely under construction and work in progress.

jfgo20, 2020-03-06:
OK. There's no rush on this on our side but it's good to know that it exists for the future.

I saw on the Impulse web site that they are talking about this (https://www.toem.de/index.php/news/214-impulse-meets-the-yakindu-statechart-tools). So, right now it's only integrated to Impulse?

rherrmannr, 2020-03-06:
I was not involved in this project. @terfloth could you please help out?

rherrmannr, 2020-06-22:
YET is now available for C in labs and can be added via the example wizard.

closed 2020-06-22 by rherrmannr