itemisCREATE/statecharts #3093: [Generator Model Wizard] use variables for targetProject and folder

andreasmuelder, 2020-05-19:
instead of generating this:

```
GeneratorModel for yakindu::scxml {

	statechart AlwaysOncycle {

		feature Outlet {
			targetProject = "com.yakindu.sct.generator.tests"
			targetFolder =  "src-gen"
		}
	}

	statechart BitExpressions {

		feature Outlet {
			targetProject = "com.yakindu.sct.generator.tests"
			targetFolder =  "src-gen"
		}
	}
}
```

the wizard should generate this: 

```
GeneratorModel for yakindu::scxml {
	var targetProject : string = "com.yakindu.sct.generator.tests"
	var targetFolder : string = "src-gen"

	statechart AlwaysOncycle {

		feature Outlet {
			targetProject = targetProject
			targetFolder = targetFolder
		}
	}

	statechart BitExpressions {

		feature Outlet {
			targetProject = targetProject
			targetFolder = targetFolder
		}
	}
}
```

tkutz, 2020-06-24:
`targetProject = targetProject` looks strange to me, should rename the variable. Also, if we can use `const` in the gen model, we should use it here.

Something like this:
```
const PROJECT : string = "com.yakindu.sct.generator.tests"
const FOLDER : string = "src-gen"
statechart AlwaysOncycle {

	feature Outlet {
		targetProject = PROJECT
		targetFolder = FOLDER
	}
}
```

Shall this be always generated, or only with multiple statecharts are selected?

andreasmuelder, 2020-06-24:
it should always be generated I think