itemisCREATE/statecharts #3134: Unused function generated for composite states without inner states

tkutz, 2020-06-24:
The C code generator generates code with [Wunused-function] and [Wunused-value] issues for the following statechart:

[image]

[image]

tkutz, 2020-06-24:
Example Models:

[Archiv.zip](https://github.com/Yakindu/statecharts/files/4825739/Archiv.zip)

terfloth, 2020-07-03:
How does this relate to #3086

closed 2020-07-03 by terfloth

tkutz, 2020-07-03:
It does not relate much to #3086, which is specific to time events on composite states while this one is about the outgoing transition of a composite state.

PR #3153 (2020-07-03): Do only generate reachable effect functions
Fixes #3134 
Fixes #3149