itemisCREATE/statecharts #3174: Workaround comments exist in releng/org.yakindu.sct.releng/pom.xml file while the related issues are already fixed.

markxiaotao, 2020-07-13:
In a research project, we analyzed the build file of repositories looking for comments with ready-to-be-addressed SATDs (self-admitted technical debt) that could be addressed. If the underlying issue is already resolved, the related code or comment can be removed or fixed. As we found an instance of this kind of asynchrony in lreleng/org.yakindu.sct.releng/pom.xml, we decided to report the issue to contribute community.
Comments in releng/org.yakindu.sct.releng/pom.xml file claim that they are workaround for https://github.com/eclipse/xtext/issues/1231, and this issue has been closed long ago, they also provided new solution: https://github.com/eclipse/xtext/issues/1231#issuecomment-460184629.

Comments location:
https://github.com/Yakindu/statecharts/blob/75ce8fc3c019d195344d0bbcdd1fd386c6809ccf/releng/org.yakindu.sct.releng/pom.xml#L213
https://github.com/Yakindu/statecharts/blob/75ce8fc3c019d195344d0bbcdd1fd386c6809ccf/releng/org.yakindu.sct.releng/pom.xml#L251
https://github.com/Yakindu/statecharts/blob/75ce8fc3c019d195344d0bbcdd1fd386c6809ccf/releng/org.yakindu.sct.releng/pom.xml#L322
https://github.com/Yakindu/statecharts/blob/75ce8fc3c019d195344d0bbcdd1fd386c6809ccf/releng/org.yakindu.sct.releng/pom.xml#L355

andreasmuelder, 2020-07-13:
Since we support multiple version of xtext (2.14 - 2.21) we can not remove the workaround here for the profiles that use older targets. 
Anyway, good job keep up with your research project 👍

closed 2020-07-13 by andreasmuelder

markxiaotao, 2020-07-13:
Ok I understand the situation. Thank you for prompt review!