itemisCREATE/statecharts #3204: Domain com.yakindu.domain.scxml is not available SCXML export bug

IOTeule, 2020-09-11:
Trying to generate SCXML file from example **TrafficLightStateMachine**.
The Editor complains about missing domain com.yakindu.domain.scxml.

Version: 4.0.0

Documentation not up to date, there is no **model.sgen** file
[image]

Trying to run **TrafficLightStateMachine.sgen** instead failed.

[image]

Export not possible
[image]

[image]

andreasmuelder, 2020-09-11:
@IOTeule thanks for the bug report. The SCXML Domain is not part of the current beta 4 release, you can download the official release 3.5.13 for the SCXML generator.

closed 2020-09-11 by andreasmuelder

IOTeule, 2020-09-11:
Danke für den schnellen Support