itemisCREATE/statecharts #3211: Missing documentation of behavior of events in guards

z3rone, 2020-10-22:
In version 3.5.9 it is possible to write events in transition guards. Doing so results in the transition being activated if the source state is entered because of that event being raised. This behavior seems not to be documented in the language reference.

Further more the behavior conflicts with the UML behavioral state machine definition. As states are stable vertices, the transitioning process should stop in a (simple) state. Further transitioning would require a new event being dispatched. However, the previous event would be consumed at this point, so the guarded transition should not become enabled.

Example of the behavior:
[image]

When event `a` is fired (a single time) the transitioning continues up to state `C`.

tkutz, 2020-10-23:
This is indeed incorrect. When event `a` is only fired one single time, state `B` should be activated.

This is already fixed in the latest version. You might want to update. 

Note, that we have released a new version 4 of Statechart Tools. But you cannot update directly from 3.x. to 4.x, you need to download version 4. However, your 3.x models should also work with version 4 (check our migration guide for more details).

Regarding the documentation you are right, it is missing there.

tkutz, 2020-10-23:
I created an issue for the documentation part in our internal tracker.

I will close this one here, as the actual behavior is already fixed.

closed 2020-10-23 by tkutz