itemisCREATE/statecharts #3214: Python code generator doesn't support hyphens in folder names

bbpaulvanharen, 2020-11-11:
The default .sgen file for c code generation uses a line 
`const FOLDER : string = "src-gen"`

The same line in the .sgen file for a python project reports an invalid name for the folder. Hence, the project won't compile

tkutz, 2020-11-12:
This is already fixed in version 4.0.2.

bbpaulvanharen, 2020-11-16:
I just upgraded to version 4.0.3 and the problem still persists. 
Noteworthy details: 
- version 4.0.3 is not mentioned in the release notes
- the pathname issue is not mentioned in the release notes

tkutz, 2020-11-16:
4.0.3 is freshly released. Release notes will be updated asap.

The default sgen for the python generator should not contain a hyphen anymore. But hyphens are still not allowed, as they are not allowed for package names.

closed 2020-11-16 by bbpaulvanharen